wlwatkins/ESPBug #5: language.h error

xndrx02, 2020-04-13:
Arduino: 1.8.12 (Windows Store 1.8.33.0) (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 160 MHz, Flash, Disabled (new can abort), All SSL ciphers (most compatible), 4MB (FS:3MB OTA:~512KB), v2 Lower Memory, Disabled, None, All Flash Contents, 115200"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\Jan Ruiz\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\Jan Ruiz\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\Jan Ruiz\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=160,vt=flash,exception=disabled,ssl=all,eesz=4M3M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=all,baud=115200 -vid-pid=10C4_EA60 -ide-version=10812 -build-path C:\Users\JANRUI~1\AppData\Local\Temp\arduino_build_834966 -warnings=none -build-cache C:\Users\JANRUI~1\AppData\Local\Temp\arduino_cache_942353 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.mklittlefs-2.5.0-4-69bd9e6.path=C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.python3.path=C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -verbose C:\Users\Jan Ruiz\AppData\Local\Temp\Temp1_ESPBug-master.zip\ESPBug-master\espbug\espbug.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\Jan Ruiz\Documents\ArduinoData\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\Jan Ruiz\Documents\ArduinoData\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.33.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\Jan Ruiz\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=160,vt=flash,exception=disabled,ssl=all,eesz=4M3M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=all,baud=115200 -vid-pid=10C4_EA60 -ide-version=10812 -build-path C:\Users\JANRUI~1\AppData\Local\Temp\arduino_build_834966 -warnings=none -build-cache C:\Users\JANRUI~1\AppData\Local\Temp\arduino_cache_942353 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.mklittlefs-2.5.0-4-69bd9e6.path=C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\tools\mklittlefs\2.5.0-4-69bd9e6 -prefs=runtime.tools.python3.path=C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -verbose C:\Users\Jan Ruiz\AppData\Local\Temp\Temp1_ESPBug-master.zip\ESPBug-master\espbug\espbug.ino
Using board 'nodemcuv2' from platform in folder: C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.0
Using core 'esp8266' from platform in folder: C:\Users\Jan Ruiz\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.6.0
"C:\\Users\\Jan Ruiz\\Documents\\ArduinoData\\packages\\esp8266\\tools\\python3\\3.7.2-post1/python3" "C:\\Users\\Jan Ruiz\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.6.0/tools/makecorever.py" --build_path "C:\\Users\\JANRUI~1\\AppData\\Local\\Temp\\arduino_build_834966" --platform_path "C:\\Users\\Jan Ruiz\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.6.0" --version unix-2.6.0
Detecting libraries used...
"C:\\Users\\Jan Ruiz\\Documents\\ArduinoData\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Jan Ruiz\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.6.0/tools/sdk/include" "-IC:\\Users\\Jan Ruiz\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.6.0/tools/sdk/lwip2/include" "-IC:\\Users\\Jan Ruiz\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.6.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\JANRUI~1\\AppData\\Local\\Temp\\arduino_build_834966/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -DNEW_OOM_ABORT -w -x c++ -E -CC -DNONOSDK22x_191024=1 -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10812 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\Jan Ruiz\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.6.0\\cores\\esp8266" "-IC:\\Users\\Jan Ruiz\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.6.0\\variants\\nodemcu" "C:\\Users\\JANRUI~1\\AppData\\Local\\Temp\\arduino_build_834966\\sketch\\espbug.ino.cpp" -o nul
Alternatives for ArduinoJson.h: [ArduinoJson@5.13.5]
ResolveLibrary(ArduinoJson.h)
  -> candidates: [ArduinoJson@5.13.5]
"C:\\Users\\Jan Ruiz\\Documents\\ArduinoData\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Jan Ruiz\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.6.0/tools/sdk/include" "-IC:\\Users\\Jan Ruiz\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.6.0/tools/sdk/lwip2/include" "-IC:\\Users\\Jan Ruiz\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.6.0/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\JANRUI~1\\AppData\\Local\\Temp\\arduino_build_834966/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -DNEW_OOM_ABORT -w -x c++ -E -CC -DNONOSDK22x_191024=1 -DF_CPU=160000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10812 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\Jan Ruiz\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.6.0\\cores\\esp8266" "-IC:\\Users\\Jan Ruiz\\Documents\\ArduinoData\\packages\\esp8266\\hardware\\esp8266\\2.6.0\\variants\\nodemcu" "-IC:\\Users\\Jan Ruiz\\Documents\\Arduino\\libraries\\ArduinoJson\\src" "C:\\Users\\JANRUI~1\\AppData\\Local\\Temp\\arduino_build_834966\\sketch\\espbug.ino.cpp" -o nul
espbug:5:23: fatal error: languages.h: No such file or directory
Alternatives for languages.h: []

 #include "languages.h"
ResolveLibrary(languages.h)

                       ^
  -> candidates: []

compilation terminated.

Using library ArduinoJson at version 5.13.5 in folder: C:\Users\Jan Ruiz\Documents\Arduino\libraries\ArduinoJson 
exit status 1
languages.h: No such file or directory

zazman, 2020-05-06:
Yeah, same error for.

willmendil, 2020-05-20:
[repost #1 ](https://github.com/willmendil/ESPBug/issues/1)

closed 2020-05-20 by willmendil