wlwatkins/ESPBug #21: I want English phishing pages

MALW4R3, 2020-08-10:
I need a file bin It is an English webpage, do you have it?

willmendil, 2020-08-10:
no, but you can make it

MALW4R3, 2020-08-10:
> no, but you can make it

I tried it and it couldn't be changed to English.

willmendil, 2020-08-10:
Well show me your work and I might be able to help you.

MALW4R3, 2020-08-11:
> 

> and

https://iot-coding-academy.web.app/file%20Download/EvilBug%20V.2%20PreRelease.bin

I tried creating it myself but it still can't change any language.

Wasim7863, 2020-09-26:
Hey I Also Tried Converting webpage into english version it isnt working .I tried so man times but cant make it work.I am doing everything as instructed into videos but all and all its loading the france version language.

mandabenjamin, 2020-12-06:
Can i captive portal by upload html?

DalenVigil, 2021-04-10:
Followed the tutorial and the video. captive.html is corrected as I'd like. However, when trying to run the python script in Terminal I get the below. I can't replace any of the info in the sketch to finish swapping over to the called captive.html.

> Traceback (most recent call last):
File "/Users/REDACTED/Desktop/ESPBug/web_converter/webConverter.py", line 8, in
from css_html_js_minify.minify import process_single_html_file, process_single_js_file, process_single_css_file
File "/Users/REDACTED/Desktop/ESPBug/web_converter/css_html_js_minify/init.py", line 13, in
from .minify import (process_single_html_file, process_single_js_file,
File "/Users/REDACTED/Desktop/ESPBug/web_converter/css_html_js_minify/minify.py", line 30, in
from anglerfish import (check_encoding, check_folder, make_logger,
File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/anglerfish/init.py", line 32, in
from anglerfish.get_clipboard import get_clipboard # noqa
File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/anglerfish/get_clipboard.py", line 18, in
Clipboard = NamedTuple("Clipboard", fields=(("copy", callable),
File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/typing.py", line 1875, in NamedTuple
return _make_nmtuple(typename, fields, module=module)
File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/typing.py", line 1800, in _make_nmtuple
types = {n: _type_check(t, f"field {n} annotation must be a type")
File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/typing.py", line 1800, in
types = {n: _type_check(t, f"field {n} annotation must be a type")
File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/typing.py", line 157, in _type_check
raise TypeError(f"{msg} Got {arg!r:.100}.")
TypeError: field fields annotation must be a type Got (('copy', ), ('paste', )).
> 

MacOS Catalina, Anglerfish installed correctly, but I'm new to this stuff, this is my first project, so I'm not sure what's screwy or where to start to fix it. My output after running the python script looks nothing like what's shown in the video, so I can't complete the replacement part. Is it possible just to convert the captive.html info to hex and directly replace it in the sketch instead of going through the script?

Secondarily, it flashes to the hardware alright, the espbug AP part works correctly, I can edit settings without issue. When I connect to the phishing page, the login page loads the original french (assuming because the above issue), I put in test user and pass, and the network disappears as it should. When I log back in to the espbug network, it shows that a connection has been made, but under the "phishing" tab, it says "NO." I'm not sure if they're both related, but I'd like to get the english page issue sorted first.

mandabenjamin, 2021-04-11:
can it run on windows 10?
as far as I know it uses linux
but I only have a computer with windows 10
is there a video tutorial

rivetin, 2021-05-19:
> Followed the tutorial and the video. captive.html is corrected as I'd like. However, when trying to run the python script in Terminal I get the below. I can't replace any of the info in the sketch [...]

try python3 insted of python