wlwatkins/ESPBug #26: Problem with updating the espbug.ino file in Arduino.IDE

K0nd0ru, 2020-11-08:
Hello...
I have a problem with updating the .ino file to implement the binary file to the esp board ... I tried by all means, but Arduino did not want to work with me ... Can anyone share the ready binary file for the board?

willmendil, 2020-11-11:
this is not for arduino, but esp boards

K0nd0ru, 2020-11-11:
I know that this is for esp tiles, but you said yourself that in order for the binary file to work, I should edit it in the Arduino.IDE program in some way
example:"To get this code running on a nodeMCU like board - such as the one illustrated below - you need to install the Arduino IDE (! I have only tried on Linux, but this should work on any OS - virtual machine included)."

willmendil, 2020-11-11:
i use the arduino ide on a vm as well. You should be able to upload the binary. but not an arduino board. 
https://github.com/willmendil/ESPBug/releases/tag/0.1

K0nd0ru, 2020-11-11:
"tiles = board" my mistake Thank you for your help

closed 2020-11-11 by K0nd0ru

K0nd0ru, 2020-11-11:
I have implemented a binary file to the board. everything works. Thanks again for your help