wlwatkins/ESPBug #27: Traceback (most recent call last)

ZainAlSheikh, 2020-11-11:
When i enter Python3 webConverter.py
it shows Traceback (most recent call last).
plz help me.

willmendil, 2020-11-11:
could you show me more of the error?

ZainAlSheikh, 2020-11-12:
No more errors.
i have Kali linux , Python 3.6.8 , pip and anglerfish installed. 
i copied web_pages folder to webConverter folder. But when i enter command.
python3 webConverter.py
it shows.
Traceback (most recent call last)
line 26 in <module>
os.kdr((stdr))
Filenotfound (errno 2)

something like that appears

ZainAlSheikh, 2020-11-12:
plz help me to solve this

priywardhan, 2022-03-12:
This issue is not solved still please help me too

priywardhan, 2022-03-12:
the error:
┌──(kali㉿kali)-[~/Desktop/ESPBug/web_converter]
└─$ python webConverter.py
Traceback (most recent call last):
  File "/home/kali/Desktop/ESPBug/web_converter/webConverter.py", line 8, in <module>
    from css_html_js_minify.minify import process_single_html_file, process_single_js_file, process_single_css_file
  File "/home/kali/Desktop/ESPBug/web_converter/css_html_js_minify/__init__.py", line 13, in <module>
    from .minify import (process_single_html_file, process_single_js_file,
  File "/home/kali/Desktop/ESPBug/web_converter/css_html_js_minify/minify.py", line 30, in <module>
    from anglerfish import (check_encoding, check_folder, make_logger,
  File "/home/kali/.local/lib/python3.9/site-packages/anglerfish/__init__.py", line 32, in <module>
    from anglerfish.get_clipboard import get_clipboard  # noqa
  File "/home/kali/.local/lib/python3.9/site-packages/anglerfish/get_clipboard.py", line 18, in <module>
    Clipboard = NamedTuple("Clipboard", fields=(("copy", callable),
  File "/usr/lib/python3.9/typing.py", line 1922, in NamedTuple
    return _make_nmtuple(typename, fields, module=module)
  File "/usr/lib/python3.9/typing.py", line 1847, in _make_nmtuple
    types = {n: _type_check(t, f"field {n} annotation must be a type")
  File "/usr/lib/python3.9/typing.py", line 1847, in <dictcomp>
    types = {n: _type_check(t, f"field {n} annotation must be a type")
  File "/usr/lib/python3.9/typing.py", line 166, in _type_check
    raise TypeError(f"{msg} Got {arg!r:.100}.")
TypeError: field fields annotation must be a type Got (('copy', <built-in function callable>), ('paste', <built-in function callable>)).

viplameo, 2022-12-19:
install python 3.7 to fix it